lshachar/Arduino_Fanatec_Wheel #2: Made the connector with an Nano, Level Shifter, the schematic and code, but no good

Jonbakhol, 2019-10-10:
@lshachar 

Hi Ishachar, I used the connection schema (the good one, not mine)
I soldered all wires correctly to the Level Shifter and to the pins. I uploaded the sketch arduino_fanatec_wheel.ino provided on https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/arduino_fanatec_wheel.ino

And plugged it in the wheelbase. But no ClubSport Porshe 918 RSR is shown.

Is there something else I have to do? Press the reset button on the Nano or something else?

I also have an Teensy 3.2 with the Teensy code, and that one is working perfectly.

Any help would be appreciated.

Thanks
JonBakhol

lshachar, 2019-10-10:
Hi! no reset is needed. Do you have both pins 3 & 5 connected to ground?
If you have a logic analizer, or a scope see if you get clock signal on fanatec's pin 8. (or simply attach an LED and a resistor to HV4 on the level shifter)

Jonbakhol, 2019-10-10:
Thanks for coming back so quick.

Yes I connected them both to ground. Measured with an ohm meter all pins to be sure no short-circuits are there (accept pin 3 & 5).
Scopes etc I  don't have. I do have some LED's and resistors. Between which point do I need to check it? HV4 and ?

Few more (maybe stupid) questions:
Do I need to connect the modified USB cable to it? (without the 5V cable)
Could it be the cheap Chinese Nano?
Do I need to put in the missing pins 9 till 13 (even there is no connection between the Nano and the pins)

Thanks for now

lshachar, 2019-10-10:
(Alright this got messy because I didn't finish my initial reply but pressed comment by mistake.)

Hi! no reset is needed. Do you have both pins 3 & 5 connected to ground?
try plugging your arduino to the PC by usb. It might get the thing to recognize if your grounds / +5v lines aren't connected well.
If you have a logic analizer, or a scope, test your logic shifter. you are suppose to get information coming in from the base on CS, MOSI, CLK lines. the arduino should send out data on MISO line.

Last thing... could be interference... Darknao said he had issues with interference so I had a ground wire twisted around each data line. I don't think it's likely to solve your problem and I hardly think it was necessary to twist some ground lines.

_____________________
answering your 2nd comment
> Do I need to connect the modified USB cable to it? (without the 5V cable)

well, The simple answer is - preferrably yes.
or at least, disconnect fanatec's pin #4  (5v supply) from the arduino, before you connect the arduino to the PC by USB.
This way, you're making sure that the arduino gets voltage from the computer and you don't connect the two power supplies to each other.
However, I definitely forgot to do that myself several times and nothing went up in smoke.
In any case you really want to get the arduino connected to the PC because you need to open the serial monitor on the PC and tell me what you see there!

> Could it be the cheap Chinese Nano?

Yes, that's what I use.

> Do I need to put in the missing pins 9 till 13 (even there is no connection between the Nano and the pins)

nope. you don't need to do anything with them.

> Thanks for coming back so quick.

lucky you.. I was on the computer :)

> Scopes etc I don't have. I do have some LED's and resistors. Between which point do I need to check it? HV4 and ?

... and ground. This is kind of long to explain, I basically offered you to check that the logic shifter was working properly, with a single LED, which isn't the best test. so I take it back, it will be difficult for you to understand if there is data transmission on the line with a single LED. instead you can test that the logic shifter is working properly. By raising one side of the logic shifter (LV# or HV#) to high - the other side should also go high. but each side will only output the voltage that it gets on its power supply pin (marked by LV or HV on the schematics, next to the GND pin)
so, if you give +5v and ground to the HV side, and you give +3v and ground to the LV side, and then you give, for instance, +3v to LV1, then HV1 should go up to +5v. you can test it with a multimeter. if you have the project right now on a breadboard, this should be a really simple test to do..
So, you are trying to debug electronically. but try the serial monitor first.

Jonbakhol, 2019-10-11:
Goodmorning @lshachar ,

I connected the Arduino to the PC with an intact USB cable (so no 5V cable cut) and checked with the multimeter the currents on the Level Shifter, on LV side and on HV side.
These are the values:

              LV side                                                                                HV side

GND + LV1 (pin 6) = 2,28 V                                                     GND + HV1 = 2,11 V
GND = LV2 (pin 1) = 3,26 V                                                     GND = HV2 = 4,66 V
GND = LV   (pin 7) = 3,26 V                                                     GND = HV   = 4,66 V
GND = LV3 (pin 2) = 3,26 V                                                     GND = HV3 = 4,66 V
GND = LV4 (pin 8) = 3,26 V                                                     GND = HV4 = 4,66 V

Except pin 6 (LV1 and HV1) all seems fine, or isn't it?
What could be the problem that pin 6 (LV1 and HV1) is that low?

Can I try to change pin6 (which is connected to D13) to e.q. D10 and update the sketch?

Many thanks again.

Jonbakhol

devActive527, 2019-10-24:
@lshachar ,

Hi. I don't know if it will help in some way, but I noticed that when I connect the 5v power supply to the arduino shortly after turning on the base there are no crc errors, while when I do it too early or too late there are errors. I discovered this while reading the com port from arduino. Maybe this is the reason for the buttons not working? Simply 5v power supply to the steering wheel should appear with some delay, and in our connection is given immediately after switching on. I do not know how it relates to the communication between the base and the steering wheel is just my theory. Can you verify it somehow?

lshachar, 2019-11-16:
CRC errors means you have bad wiring / bad ground / noise issue. Since the button delay problem happens even when there are no CRC errors - the two aren't related.

lshachar, 2020-04-28:
[The new diagram](https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png) and fixes I made to the SPI communication, means everything works much better now.