lshachar/Arduino_Fanatec_Wheel #4: DIY arduino steering wheel to fanatec csl elite wheelbase

Pistolidas, 2019-11-16:
Hello All,

Has enyone been able to get either the Teensy 3.2 or the Arduino Nano working on a Fanatec CLS Elite V1?
I tried both, but no succes. :(

Regards,

Pistolidas

MagnusThome, 2019-12-16:
I have it working on a CSL Elite+ base

https://rejsa.nu/forum/viewtopic.php?p=1588680#1588680

Pistolidas, 2019-12-16:
Hello Magnus,

Thnaks for your message. Would you be willing to share your connections and arduino sketch?
Are you using a teensy 3.2?

Looking forward to your reply.

Regards,

Pistolidas

MagnusThome, 2019-12-17:
Hi!

 

 

I use an Arduino Nano with his code just as is with no changes:

https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/arduino_fanatec_wheel.ino

 

 

I also connected everything 100% according to his diagram:

https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

 

 

It works right away. I did almost make a mistake though connecting the wires to the home made connector going into the base unit’s axle connector. I misread the connection diagram mirrored, to be viewed as you solder the pins towards the base unit, but it’s viewed in the picture from the base outwards. Just double, triple, quadruple check all the wiring is my tip.

 

Email me again if you have any more questions!

 

 

 

/Magnus

Pistolidas, 2019-12-18:
Hello Magnus,

Thanks for very much for your reply. 
I also made this unit and it works on a Clusport V1 wheelbase but not on my Elite V1 wheelbase. Would you know which firmware version and which driver version you have installed for your CSL Elite wheel base.

Kind regards,  

Pistolidas

MagnusThome, 2019-12-18:
The Fanatec propertiy page in Windows says:

PC driver: 328
WB FW: 630

But I've found a problem, depending on what Aduino board I use it works or not.

With the working board:
- When I plug it in to the base the base unit's red power led fades down and up once and then the Clubsport wheel shows up on the windows property page
- All four data pins have signals, measured with a scope

The not working board:
- When I plug it in to the base the base unit's red power led fades down and up coninously. The Clubsport wheel never shows up on the windows property page
- Three data pins have signals, measured with a scope. They look like on the working board but they are there for a second, gone for second, there for a second, and so on.
- D12 is low all the time. No signal visible at all like on the working board. Tried a 4K7 pull up resistor to 3V3 but no change.

lshachar, 2019-12-26:
> I also made this unit and it works on a Clusport V1 wheelbase but not on my Elite V1 wheelbase. 
> Pistolidas

Does your Elite V1 Wheelbase works with a different wheel?
In case you haven't tried - I suppose that the wheelbase is at fault
If you have tried and it works - I'm really surprised! Follow MagnusThome's advice: 'Just double, triple, quadruple check all the wiring is my tip'

MagnusThome, 2019-12-26:
Pistolidas, did you connect the two ground pins to each other? Its a long shot but might affect the function a bit differently on different base version, who knows...

lshachar, 2019-12-27:
> Pistolidas, did you connect the two ground pins to each other? Its a long shot but might affect the function a bit differently on different base version, who knows...
It actually isn't a long shot, without the two grounds connected, my CSW V2.5 will not recognize anything. it must be connected in order to work.

Pistolidas, 2019-12-27:
Hello Magnus and Ishachar, thanks for your help. I ordered a new Arduino Nano and I'll give this a try this weekend.
With the "old" Nano I have connected the GND to both pins 3 and 5 of the round connector to the wheel base, as described here: 

https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

I double checked the connections ones more and all seems to be okay.

I have two wheels for my Elite wheel base the Fanatec XBOX P1 and the Fanatec McLaren GT3 both work properly. 

I have also tried the Teensy 3.2 setup as described here: https://github.com/darknao/btClubSportWheel/issues/12

This also only worked on a Clubsport V1 and not on my Elite V1. 

I hope the new Nano will work but I have my doubts, to be continued.......

Pistolidas, 2019-12-29:
It worked, I replaced the Aruino Nano and it recognized it as the Clubsport Porsche 918 RSR wheel, Force Feedback works.
Thanks for your help!!

DavDaddy, 2020-12-13:
> 
> 
> It worked, I replaced the Aruino Nano and it recognized it as the Clubsport Porsche 918 RSR wheel, Force Feedback works.
> Thanks for your help!!

Not to hijack this thread but does anyone have any idea if this will work with the ClubSport v1 wheelbase?

I know the Porsche 918 wheel is compatible so I figure it should but I wanted to double check before I rip my DIY HOTAS apart?

Thanks.