lshachar/Arduino_Fanatec_Wheel #5: Adapting to Clubsport V2.5

matt0hait, 2019-12-21:
I'm working on adapting this to my Clubsport V2.5 using an ESP32. I'm a little stumped on the pinout. It seems different from the one provided in the repository. I thought that the Fanatec rims were compatible between the bases. I'm limited in my options since I don't have a Fanatec rim yet. This is what I found so far by probing the base's PCB:

[image]

Anyone else take a crack at this yet?

MagnusThome, 2019-12-22:
I'm a bit confused with your pinout numbering. This is the original connector numbering?

The four uncommented pins are 5V, GND, +D, -D

[image]

MagnusThome, 2019-12-22:
This pinout works on a CSL Elite + I have here and it defintely works with the same steering wheels as with your Clubsport V2.5

matt0hait, 2019-12-22:
Just re-probed with my oscilloscope, (using the pin map you provided) I have 5V on pin 11 &13. Pin 4 definitely connects to ground for me. I was able to continuity check it to a Vss pin. Is it possible that the rims provide 5V to the base through pin 4?

I'll have some level converters later this week. Saw some diagnostic pins on the inside of the base. Gonna see if they can spill the beans.

matt0hait, 2019-12-26:
(Using your pin map again) Pin 3 does output 3.3V, but once it is pulled to ground the signals become alive. I'm still not receiving any signals on MISO or MOSI. Using the serial interface on the arduino doesn't create any signals either. Think I'm going to need to try another arduino.
[image]

lshachar, 2019-12-26:
When I initially tried to implement this I wanted to go the ESP32 route, but as far as I can remember - nobody was able to put it in ISP Slave mode. Even though some documentations were referring that.
I advise you to try getting your ESP to receive data as an ISP slave, before attempting to use the code.
(Actually, I would advise you to get it working on an arduino first... would probably make your life much easier)

danieleibarra, 2019-12-28:
Has anyone already found the solution to go on a v2.5 engine?

lshachar, 2019-12-28:
> Has anyone already found the solution to go on a v2.5 engine?
I just changed the title, to explain the the OP is trying to adapt to an ESP32 from an Arduino;
My project is based on an arduino and a CSW V2.5. So to answer your question - yes, there is a solution for a V2.5...

lshachar, 2019-12-28:
>Pin 4 definitely connects to ground for me. I was able to continuity check it to a Vss pin. Is it possible that the rims provide 5V to the base through pin 4?
>Pin 3 does output 3.3V, but once it is pulled to ground the signals become alive.

I'm pretty sure you're reading the connector mirrored. the picture I gave is the side *on the rim*, wheres the side you're probing is the connector side *in the wheelbase*.

> Is it possible that the rims provide 5V to the base through pin 4?

The rim gets 5V from the wheelbase, not the other way around.

danieleibarra, 2019-12-28:
> > Has anyone already found the solution to go on a v2.5 engine?
> 
> I just changed the title of this issue, to explain that the OP is trying to adapt to an ESP32 from an Arduino;
> My project is based on an arduino and a CSW V2.5. So to answer your question - yes, there is a solution for a V2.5...

Ah that's fine, so how would the montage be? because I would like to have it

matt0hait, 2019-12-28:
That picture of the oscilloscope is with an Arduino Leonardo.Still no dice. Going to pick up another Arduino soon and see if that fixes the issue.

danieleibarra, 2019-12-29:
> That picture of the oscilloscope is with an Arduino Leonardo.Still no dice. Going to pick up another Arduino soon and see if that fixes the issue.
 

Is the assembly but not yet confirmed to work?

lshachar, 2019-12-30:
@danieleibarra , I can't understand your English.

danieleibarra, 2019-12-30:
> @danieleibarra , I can't understand your English.

I mean if there is already something that works 100%

lshachar, 2019-12-30:
I keep telling you **YES**. the project is fully working with an arduino and many fanatec wheelbases. including CSW V2.5.

danieleibarra, 2020-01-06:
Ok I have the sketch for adolescence 3.2 I just don't know where they would go, help me

lshachar, 2020-01-08:
not befoe you could explain in English what your issue is.