lshachar/Arduino_Fanatec_Wheel #8: Startup delay

MagnusThome, 2019-12-31:
When I turn on my CSL Elite + base unit most of the time the Arduino/wheel isn't detected and I have to restart the Arduino with it's reset button after some seconds after turning everything on, then all is well.

I guess it's because the Arduino is booting to fast compared to the wheel base? So I added a one second delay right at the beginning of setup() and now the Arduino/Wheel is detected properly _every_ time I turn on the base unit.

lshachar, 2020-02-10:
That's weird, I'm guessing it's an issue with the SPI interface, as the arduino is always looking for the start of a sequence sent by the wheelbase, and then it just sends off the same message out. I'm going to blindly add your delay suggestion to setup() (because on my csw this was never an issue)
Thanks!

closed 2020-02-10 by lshachar

sabifa, 2020-02-14:
Hey! I know this issue is closed already but today I received my quick release and built the circuit on a breadboard to find out that its not working.
I had to press the reset button on my nano a couple of times and it only worked like 3 / 10 times.

After some digging in the code I thought that I should try to remove the delay which was added a couple of days ago. After I removed it, it worked 100% of the times I've tried! (5/5)

It would be nice if someone else could also try this out so we can be sure if its needed or not.

My specs:
CSL Elite + Wheelbase
Wheelbase Firmware 662
PC Driver 346

MagnusThome, 2020-02-14:
Interesting! With my nano board it's exactly the opposite. I think it's all
down to the general problems to get a stable communication even though all
my wires are only 2-3 cm long. Very odd.

lshachar, 2020-02-14:
Hmm.. thank you @sabifa!

lshachar, 2020-04-28:
Those issues were likely because of bugs that I recently fixed in the spi communication...