lshachar/Arduino_Fanatec_Wheel #9: Adapting to Arduino Pro Micro 3.3V 8 Mhz

sabifa, 2020-02-17:
Hey!

I've tested this code using an Arduino Nano + Level Shifter and it worked like a charm! As I want to design the smallest possible PCB so that it fits into my Fanatec QR Adapter (52mm to 70mm) I want to use a 3.3V powered Arduino so I don't have to use a level shifter which is quite big.

@lshachar stated that he didn't try a 3.3V Arduino yet, because he thinks that it will be a problem later on if we want to use the buttons. As the button functionality isn't yet fully implemented and I'm planning to use something else this won't bother me. I just need the base to think that a Fanatec rim is attached to it! 😅 

I only found a few 3.3V powered Arduino's and made my choice for the Pro Micro. 

I have changed the ``CS_ISR`` pin to 3, as pin 3 maps to interrup 0 (INT0) on the board. More info about the board can be obtained [here](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide/all#hardware-overview-pro-micro).

I've wired everything up and connected it to my base, but it instantly disconnects from my pc and reboots (the rev lights light up and it spins to center itself). So there is definitely something not working as it should.

The strange thing is, if I disconnect pin 4 from the Fanatec QR (5V) and connect the pro micro to my pc via usb to have a look at the serialport, my base **does not reconnect** if I connect the arduino to it 🤔 
The serialport only says that "incoming data crc8 mismatch!" is wrong. I printed the MISO and MOSI buffers and they are:
```
MOSI:0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
MISO:0xA5 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x13 0x25 
```

Just in case, here is my wiring:
4 - RAW
7 - VCC
3 - GND
5 - GND
1 - 14
2 - 16
6 - 15
8 - 3

Here is the pinout of the Pro Micro as found on the Sparkfun Site:
[image]

Please let me know if I can do something else to provide more information that could help to solve the issue. Did maybe one of you already try another 3.3V Arduino, if so did it work for you? I'm not fixed with the pro micro!

Thanks!

MagnusThome, 2020-02-17:
Can the 32u4 act as SPI slave like the 328p? Curious to what choices we have to chose as controller chip.

MagnusThome, 2020-02-17:
I think the Pro Mini is 328P 3.3V and it's very small.

sabifa, 2020-02-17:
Reading [that thread](https://forum.arduino.cc/index.php?topic=255594.0) it seems like that it can not be used as a spi slave out of the box..
They wrote that they somehow grounded pin 17 (rx led) and it worked. I'm not exactly sure how to do that though. Desolder the led and just connect the ground pad to the other one?

sabifa, 2020-02-17:
> I think the Pro Mini is 328P 3.3V and it's very small.

Yes I saw the Pro Mini as well but as it doesn't have a USB port I thought it would be better to use a different chip, just to keep things as simple as possible.

sabifa, 2020-02-17:
Okay I just ordered a Pro Mini 3.3V and will report back in a couple of days when it arrives. I think the Pro Mini will be better compared to the Pro Micro as it uses the same 328 as the nano (as you already said) :sweat_smile:

lshachar, 2020-02-18:
> Okay I just ordered a Pro Mini 3.3V and will report back in a couple of days when it arrives.

I think it should work out just fine for you. :)

sabifa, 2020-02-19:
Just received the pro mini and uploaded the sketch. Unfortunately the same happens, I connect it to my base and it instantly disconnects and restarts. 
I guess we should close this issue here and start a new one for the pro mini. I will do some more testing to see if I find something whats causing the issue. Does someone else have a pro mini 3.3v as well and could try it please?

sabifa, 2020-02-22:
Did some more testing and couldn't make it work 😞 
What are your thoughts on this? Maybe it's the 8 MHz which stop it from working.

smollda, 2020-02-28:
I found this :
"Atmega328P Datasheet 19.5.2,
19.5.2 SPSR - SPI Status Register
Bit 0 - SPI2X: Double SPI Speed Bit
When this bit is written logic one the SPI speed (SCK Frequency) will be doubled when the SPI is in Master
mode (see Table 19-5). This means that the minimum SCK period will be two CPU clock periods.
When the SPI is configured as Slave, the SPI is only guaranteed to work at fosc/4 or lower.

Thus, 8 MHz transfers should not be counted on to work in Slave mode, only in Master mode."

posted on arduino forum here https://forum.arduino.cc/index.php?topic=424260.0 (post #3]

I was also looking into using a 3,3v arduino for more simple solution, but it seems it won't work.

sabifa, 2020-02-28:
Good catch! So 8 MHz is just not enough.

After some more digging about running Arduinos at 3v3 and 16 Mhz+ I found the [Adafruit Trinket M0](https://www.adafruit.com/product/3500). Judging from the specs it doesn't look too bad! 3.3V @48Mhz. (thats 3x the speed of the 328p, is there something like too fast?)

About using it as a spi slave I've found [this thread](https://forums.adafruit.com/viewtopic.php?f=25&t=94366) which doesn't contain much information unfortunately.

smollda, 2020-03-02:
I've also found this https://flytron.com/multicopter-parts/116-3v-16mhz-pro-mini-with-100-arduino-compatibility-.html , but it's a bit too expensive (for what it is). I've had arduino nano, level shifter and PCB prototype board lying around, so I'll just use that. It's not going to be pretty though ...

lshachar, 2020-04-28:
You might want to give the 3.3V arduino another go. I debugged a few things on the SPI communication, it's a lot more reliable now. [notice the changes in the diagram](https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png) (Added Diode, and D2 is connected to D10). the latter is really important for proper SPI communication and may fix the issue that you've had before.

sabifa, 2020-04-28:
I will! Thank's for your effort in this project 🚀 

Could you tell us a bit more why it's necessary to connect D2 and D10? Is it because of the Slave Select?
Sorry if I missed it but what kind of Diode do I exactly need?

lshachar, 2020-04-28:
:)
I'm using an interrupt routine on D2: Every time that the cable select pin is pulled high by the wheelbase (right after a transaction ends) the arduino gets ready for a new data transaction.

but I still had a problem where the incoming SPI data was pushed a couple of bits out of alignment. That problem would come and go. So for example the 1st bit of the 1st byte sent by the wheelbase, would sometimes register as the 2nd or 3rd bit of the 1st byte in the arduino, etc. And once that happens - all the packet is corrupted and you have nothing but CRC8 errors. 
I found out that for the serial parallel interface to work correctly, even in slave mode, it also expects the CS line to drop in the beginning of a transaction. Which also resets the single-byte SPI index, So that the next incoming bit will be registered correctly as the first bit in the buffer.

The diode is needed to separate the wheelbase 5V power supply from the computer USB 5V power supply.
I'm not sure what schematic are you using for 3.3V arduino, but I can assume that when you plug the arduino to a computer via usb, the arduino gets 3.3V from both the wheelbase and its internal 5V to 3.3V voltage regulator. If that is indeed the case, Then yes, you should add the diode.

Any old 0.5A diode should be enough. I believe the one I used is about 3A.

sabifa, 2020-04-28:
Thank you very much for that detailed explanation, it's very interesting! I also saw that the button inputs are now working correctly without the delay, that's super handy to build a custom wheel. 

I will try the 3.3V arduino again and will report back :)

sabifa, 2020-07-29:
Sorry for the long delay! 
Today I tested the new code with a regular 3.3V Pro Mini which unfortunately did not fix my problems. The same happened as before (rebase restarts, power led slowly fades from on to off).

I also purchased an [Arduino V2 Pro Mini](https://www.tinyosshop.com/index.php?route=product/product&product_id=275) which offers the possibility to run 16 MHz on 3.3V. With that it works in about 3 / 10 cases, which is not that good - but hey at least it works, kind of 😅 
In the fail cases my wheelbase restarts as with the regular 3.3V Pro Mini. 

I also noticed that the wheelbase thinks the D-Pad button is held down the whole time, did anyone of you experience that as well?

I'm not to sure what causes the issues, maybe it's the 16 MHz which aren't officially supported when running 3.3V or it's the long jumper cables connecting the Pro Mini to the Fanatec QR pins.

If I find the time I will create a small pcb to make sure the jumper cables are not a problem here and report back!

Hilpas, 2020-12-15:
Has somebody  tried it again?, I'm currently trying to get it running on my pro Micro but i got similar MISO/MOSI results as you got initially. My wheelbase isnt restarting or anything like this tough. My Wiring is the same apart from 3.3v not being connected. Unfortunately no wheel is recognised by the wheelbase. 

Is it set that the Processor frequency is not enough or is it just a Atmega328P at 3.3v thing? Has anybody got it working by now?

sabifa, 2020-12-15:
I suspect that 8 MHz is not fast enough as I couldn't make it work with a pro micro either.

Try to buy a Pro Mini V2 which is able to run 16 MHz at 3.3V and you should be good.
You can find more Infos here #13

Hilpas, 2020-12-16:
Oh havent seen that conversation. Will give the Pro Mini V2 a shot and see if that works out for me. Thank you for your investigations!

Hilpas, 2023-06-04:
I am still using it in this configuration Pro Mini V2 16Mhz 3.3V. Its working flawlessly for me and its the only wheel i use. 
I made my own Quickrelease Adapter that replicates the original Fanatec Connector. 
https://www.thingiverse.com/thing:5168903

Kuillemaul, 2023-10-08:
Where can you buy Pro Mini V2's now?