lshachar/Arduino_Fanatec_Wheel #10: Issue with turning the wheelbase off?

lshachar, 2020-02-18:
Hi guys,
I have this thing running on a CSW V2.5, and I cannot turn the wheelbase off by holding the red 'start' button. (I'm assuming it's because that the wheelbase is trying to shut the rim off before powering down, but the rim just keep repeating "Hey! I'm a rim!")

I'm wondering if anyone else has the same problem or not? and what wheelbase are you using?
Thank you.

MagnusThome, 2020-02-18:
Do you have the USB connected? It'll drive power back into the wheel.

lshachar, 2020-02-18:
usb disconnected, and I'm pushing it for even longer.
thanks for the feedback Magnus. (csl wheelbase iirc)

sabifa, 2020-02-25:
Just tried it and I am able to power it off just as I'm used to with an original Fanatec rim.
I also have the csl elite + as Magnus.

lshachar, 2020-03-24:
After reflashing the firmware (to the same, latest one) I don't have this issue any more.

closed 2020-03-24 by lshachar