lshachar/Arduino_Fanatec_Wheel #11: incoming data crc8 mismatch

smollda, 2020-03-07:
Hello, I've connected arduino nano with logic level shifter and this is the message I'm getting, when connecting to serial interface of the arduino. The wheel is acting weird, sometimes it would keep restarting itself, sometimes it works. However, driver software shows Elite steering wheel P1 connected, sometimes it's the BMW one, sometimes Porsche (not sure which type). Also it shows that all the buttons are being pressed randomly.
I have double checked the wiring, but I'll do that again.
I have even used it for a while in ACC, I just had to unmap all the fanatec buttons, as I was getting fake inputs.
Any suggestions? Could it be bad wiring issue? Or maybe bad arduino nano clone?

lshachar, 2020-03-08:
...and you're also recieving the 'incoming data crc8 mismatch' message on the serial monitor?
sounds like you have a bad connection / wrong connection / noise problem.
try twisting another wire around each data wire (mosi, miso, clk) and connecting those wires to ground.
if you can get your hands on an osciloscope, you should be able to see if the lines are noisy.
maybe it's a bad level shifter?

smollda, 2020-03-08:
Hi, thanks for the quick response! After some testing today I can say your "noise" theory is probably true. I made second testing circuit with different arduino and different level shifter and after turning on, serial monitor showed about 50% crc8 mismatch messages mixed with what looked like 3 hex numbers (I assume those were correct messages). I was also using laptop running on baterry at that time, which might have helped ... ? No buttons were being pressed randomly. The wheel appeared as a Porsche 918 RSR steering wheel.
Anyway, I'll make a new properly soldered connector and probably even custom PCB to fit arduino and level shifter and try to make some sort of shielding for data wires.
I also have access to an oscilloscope at work, so if the new circuit fails, I'll check what's going on with those data lines.
Good news is, after installing new beta driver and wheel + motor firmware and connecting the new testing circuit, the calibration issue was gone. I just had to calibrate wheel center using the new driver (version 356)

lshachar, 2020-04-28:
there should be a lot less CRC8 errors now that I fixed a few SPI communication bugs