lshachar/Arduino_Fanatec_Wheel #12: Encoders, Joysticks, and LED support

BlakePries, 2020-06-09:
I am currently working on making a DIY wheel for my Fanatec base. Looking through the current code and documentation, it appears that there is support for mapping all of the existing buttons from the Fanatec wheels. However, I don't see any mention of the encoders, joysticks, or led rev lights. Most Fanatec wheels have a "funkyswitch" which is a 4-directional d-pad/joystick that also has an encoder built in. Is it possible to add support for that input?

Also, I am currently working on 3d modeling a DIY electrical connector that uses inexpensive header pins so there isn't a need to do the molding using an existing connector,

lshachar, 2020-06-11:
Hi there,
You are correct, there's no support for LEDS / encoders / joysticks in the code. however, I feel like all the heavy lifting is already done in the code, so it should be very straightforward to implement these quite easily, if you have Arduino programming experience. good luck!

andrewkerr5, 2020-08-27:
> I am currently working on making a DIY wheel for my Fanatec base. Looking through the current code and documentation, it appears that there is support for mapping all of the existing buttons from [...]

how did you get on with the model of the connector, im thinking of doing something similar, did you find any suitable pins?

EtienneGameSeed, 2020-10-29:
Hi,
If you're interested I'm selling a board based on Teensy LC, with which can deals with buttons, encoder, joystick, RGB leds, 7 segment display and rumble motors. There's an 8 pin jst 2.0 connector so the connection to a Fanatec QR is easy
But it is not based on Ishachar code and not open source : 
https://www.electroseed.fr/shop/product_info.php?products_id=351

Fish2, 2020-11-07:
@EtienneGameSeed out of interest do the inputs and outputs go throw the Fanatec wheel base ? so my main question is will I be able to configure the leds in Fanalab ?

lshachar, 2020-11-07:
they do, and you will.

EtienneGameSeed, 2020-11-08:
Yes I confirm, it wheel behave exactly like a Fanatec wheel, and you'll even be able to configure FFB from the wheel as with a Fanatec wheel

Shane87F3Ng, 2022-03-07:
> 

Hello Ishachar,

first of all, thx very much for your work, that is just awesome!
I have one question about how to add additional buttons / encoders..
Do i need to change to another wheel than the Porsche RSR? or just modify the button matrix and bytes?

thomaswang04, 2023-12-28:
I know this is an old thread but I did some poking in the code and figured out how the joystick code works for at least the porche 918 rsr in the base code. returnData[5] is for the x axis and returnData[6] is for the Y axis. 0x00 and 0xff are center position. 0x00 to 0x7f is negative x/y where 0x7f is the farthest the joystick goes. Then 0x80 is the opposite side of the x/y axis which goes all the way back to 0xff which returns to center. 

I spent like an hour trying manually adjusting values before trying to cycle through each value and discovering how it works.

hopefully this helps @Airnautique if he is still working on this or anyone else who reads this thread.