lshachar/Arduino_Fanatec_Wheel #14: Proposed additions to the schematics

DavDaddy, 2020-12-27:
Hello I wanted to see if you'd be interested in adding a couple alternate schematic images?  I edited yours to show an Arduino Uno in place of the nano and also both using a single voltage divider on the MISO line in place of the level shifter.

If you could give them a quick look to make sure I didn't get anything wrong?  Then I can issue a pull request if you'd like to add them to the main repository here?

thanks!

[image]

[image]

[image]

lshachar, 2021-03-07:
Hi, thank you for your work!
I'm not sure if merely replacing the arduino nano with an uno helps out many users; since it only contributes by showing the physical pins location on the board. As you know, regardless of their pinout location, they are exactly the same pins. That is not surprising since an Arduino uno and a nano have the same exact functionality, allabit in a different form factor.

I'm not completely at ease about publishing exact directions about how to remove the logic level shifter, and using a voltage divider and apparently (as you have found out) stepping up the MOSI from the wheelbase to the 5v arduino input - so, I wouldn't like to add that schematic to the repository. (Since there's a high risk of feeding 5v into your expensive Fanatec wheelbase)
However, you can reattach the schematic that you put up with the voltage regulator (and transistor). If someone digs them up from here and understands the risks - I have no problem with that!

The ATMEGA328 datasheet states: digital inputs transition from Low to High happens at 0.6 * Vcc, or 3V. Now, AFAIR the Fanatec wheelbase MOSI outputs at 3.3V. (You would have to check if I actually remember that correctly!)
This means that you shouldn't have to level shift the MOSI line. Maybe check your connections and voltages? feed your arduino from a potentiometer and check in what voltage does it actually reads High?

If you still want to step up the (3.3V?) MOSI signal to 5V, you will need a non-inverting level shifter. [See the one in the first answer here](https://electronics.stackexchange.com/questions/107382/use-bjt-transistor-as-a-switch-without-inverting-the-signal)

tahir51, 2021-10-02:
Hi Ishachar. Thanks so much for sharing. I am a total noob when it comes to electronics but I will still try it. Could you tell me which diode would be best as I need to order it? I didn't want to open a new thread for this.

cihaneren, 2022-05-03:
Actually as another noob at electronics, I was looking for the correct diode for this project and I saw your question @tahir51 :) @lshachar could you please share the diode name with us ? And a huge thank you for your amazing project ! 👍

lshachar, 2022-05-03:
1N4001.
it really doesn't matter what diode you use, as long as its max current is more than 0.25W.

you're welcome :) ,
This project is donationware, if my work helped you out [maybe get me a beer! :)](https://paypal.me/lshachar?locale.x=en_US)

Alexbox364, 2023-05-20:
Hi everyone, and thank you for the amazing work you shared here.
I’m currently building my own Porsche 911 GT3 R steering wheel based on turn project : https://simracing-pc.de/en/2022/12/14/turn-racing-diy-project-911-gt3-r-2023-diy-steering-wheel/

I managed to establish SPI communication with my DD1 base and run 24 buttons. I’ll investigate using the encoder capability of SPI protocol.

what I did not manage to do is get enough power from the wheel to run the arduino. I have to hook up the arduino’s USB to the computer.

when I try to get power from the base only, it turns on for 2 second and then the base’s screen turns black.

I tried to get power from pin 4 (+5V) And 9 (USB +5V), both have same effect.

did someone manage to get power from wheelbase for the above shown schematics without any issue? 

Cheers!

Bob2345de, 2024-03-10:
Re: „ when I try to get power from the base only, it turns on for 2 second and then the base’s screen turns black.“

Many thanks for this comment. Had been trying to get this to work at first without usb connection to the Arduino. Had the same issue. With the Nano connected to usb works perfectly with my DD1. Am surprised that the Arduino’s minimal 5v draw is tripping over current. This could be the same issue causing CSL DDs to turn off when the wheel is plugged in. Will attempt to get the code running on a STM32 bluepill at 3v3.

Huge thanks to the author for his work!

lshachar, 2024-03-11:
> what I did not manage to do is get enough power from the wheel to run the arduino. I have to hook up the arduino’s USB to the computer.
>
> when I try to get power from the base only, it turns on for 2 [...]

You could try adding a capacitor on the 5v line to stabilize the initial current draw. But really you'd want a scope to see that initial current draw yourself. If it's big - there might be something wrong with your Arduino/wiring. I would also experiment with finding the tripping point for the 5v supply on your wheelbase.
I suppose you don't have a scope. 
The Arduino should not draw more than 200ma I'd think.
5v
v=i*r
r=25ohm
Use a 25ohm resistor for a 200ma current on a 5v line. Make sure it's a 5v line though. If the wheelbase is okay with that, then the problem is probably your Arduino.

Bob2345de, 2024-03-14:
So, I took apart a fanatec wheel today to wire up a logic analyser and see what is actually going on when conecting a wheel to a DD1.

As you can see in the pics below, the spi communication starts within about 200ms of connecting the wheel. Could it be that newer bases are expecting that communication to be establiched earlier that older ones did and, therefore, are shutting down when that doesn't happen in time?

From what I have been able to establish the normal arduino boot process (using the boot loader) takes approx 2000ms, ten times the expected duration. Also, when using USB power, my arduino works without issue, perhaps because it can establish the SPI cinnection quickly because it has already booted prior to attaching the wheel. Did those who have this working on newer bases flash the firmwire via ISP without a bootloader, allowing the arduino to get SPI up in time? Will tty that on the WE.

You'll notice that top 2 traces (3v3 and 5v) go high a bit later than the SPI lines. Those are the shortest pins (marginally) on my wheel, the delay varies directly with the speed I plug in the wheel, therefore, I believe, but am not certain, that's insignificant.

BTW current draw definitly isn't causing the shutdowns, both the Fanatec wheel and my arduino nano pull the same 20-30mA.

Plug in slowly:
[image]

Plug in quickly:
[image]