lshachar/Arduino_Fanatec_Wheel #16: Problem with Fanatec DD1 Base

rcalleso, 2021-12-27:
I guess this should work with DD1 base, has anyone tried it?

In my case, when I connect the arduino to the base, the displays turns off, and I have to power off and on the base to start working again. (just the base, the arduino adapter haven't worked yet)

Any clue?

lshachar, 2021-12-27:
Yes, it should work just fine with any wheelbase.
Check your wiring. When you're done, check them a couple more times...
Try to have rather short wires, and make sure there's a good connection on
all the pins.
with my CSW V2.5 I had issues where the wheelbase reset itself (turned off,
then on after a few seconds), which happened because the wheelbase wasn't
happy about the communication it got from the arduino.
It could be that the DD firmware behaves differently in the same situation.
Otherwise maybe you're shortening 3.3V or 5V into ground.

Good luck!

rcalleso, 2021-12-27:
Oh, thank you Ishachar!!!!! I was afraid you had abandoned this project :)

I've done two versions, one with connectors, and another one with short wires soldered directly to the arduino, and logic shifter.

Is there any way to be sure the arduino is doing what should be doing??

BTW, your answer is the best Christmas present for me ;)

lshachar, 2022-01-16:
Have the arduino connected to your PC via USB, and start a serial monitor.
If you see a lot of "_____________incoming data crc8 mismatch!________" messages, then your SPI connection to the wheelbase is not working properly. Mine has been like that for the longest time (my PCB was home fabricated and it's really not that good) but if some messages go through, then your wheel should get recognized by the rim and FFB will work. (however I sometimes get erratic double shifts behaviour.)

type c6 and send it (on the serial monitor). you should see a long list of Hex bytes and then "S _ 1" in a new line. this is the message you get on the alphanumeric display when you enter the menu system.
read all the comments in the source code (.ino file), should give you plenty of clues.

 enter the menu by sending c6 to the arduino.