lshachar/Arduino_Fanatec_Wheel #17: Compatibility with CSL DD

tahir51, 2022-01-04:
Hey, anybody made this project work with CSL DD? 

I couldn't get it to work. I double and triple checked the connections but the only thing happened, was the wheelbase turned off. Also got error message on the TM1637 display.

Berstrpriv, 2022-01-06:
So, I am currently also trying this with a csl dd.
My findings so far:
When I attach an P1 wheel in pc mode, then disconnect the wheel, the emulator works. I see a P1 wheel attached, also force Feedback test works.
Now the funny part: my emulator/nano is only connected with 3 wires. Both gnds and the 5v.
In your case, maybe try to set the base into the comp csw2.5 mode. Tell me, if that works cause I don't want to switch wheels everytime i start racing

FixyNeko, 2022-01-13:
Hello
I confirm this project works with a csl DD :thumbsup:
However, this base being brand new and me having no oem wheel, I did encounter some problems : the base was shipped with a version 0 firmware. After upgrading it with the fanatec driver tool, I got both the base and the wheel to be detected.

I ported this project to a RP2040 µC, onnected only one of the two grounds (the one on the outer pins) and used the 3.3V power supply pin.

Berstrpriv, 2022-01-14:
@FixyNeko Thanks for this info. I will give it another go this weekend. Did you also experience random shutdowns when connecting the emulator? Which pins did you use? Mine seem to be too thin.

tahirx51, 2022-01-14:
@Berstrpriv it's probably not the best method but what I did was using male Jumper cables. But they are too thin so you need to thicken them with a little soldering tin. And then try out and adjust until it fits perfectly without play. Also printed a custom SPI 3D Part so the jumper connectors sit tight in them without coming off. 

I will give this another try tonight and let you know if I make some progress

lshachar, 2022-01-16:
> I couldn't get it to work. I double and triple checked the connections but the only thing happened, was the wheelbase turned off. Also got error message on the TM1637 display.

what error message are you getting? if you're getting "CRC" then it means the SPI interface is receiving corrupted data on the arduino side. If no packets are transmitted 100% correctly (from the arduino to the wheelbase), then FFB effects will not be enabled on the wheelbase. (the CRC message is generated by the arduino and sent to the display)

If you're getting "UPD" message flashing, then that's better. this message is sent directly from the wheelbase, which is trying to update. This also means your SPI connection is working.
This happened to me quite a lot recently. I just shrugged it off and restarted the wheel. I'm not sure if it was stuck on trying to update the rim (trying to update the arduino code...) which isn't supported and I don't plan on supporting it, or was it just trying to update the wheelbase' and motor firmware, and I killed it mid-way. "UPD" isn't displaying anymore and I just went into fanatec firmware manager (inside Fanatec's driver page, on your PC) and everything is updated to the latest version, (also there's no update waiting for the rim.) So, this is probably not your issue, as you would've just waited or restarted and went on your way. (Your issue probably isn't the best place to write up about this "UPD" issue that I faced, which isn't mentioned anywhere else on this github. oh well).

I currently have lots of CRC8 errors on my PCB, but still able to use the wheelbase.  (see what I wrote here #16 )

(the CRC messages on the alphanumeric display can be hidden, by commenting out #define DISPLAY_CRC8_ON_ALPHANUMERIC. but really you should only do that if it's MOSTLY working fine, and seldomly, every few seconds it recieves a bad packet.)

tahirx51, 2022-01-16:
Thanks for your reply @lshachar. Yes I got the CRC Error. I wired it up a 2nd time with a new level shifter and cables just to make sure there was no faulty connection or soldering. But the outcome was the same. The wheelbase immediately disconnects as soon as I connect the SPI. Almost seems like there is a short circuit and the wheelbase is protecting itself. I did the wiring exactly like in the simplified schematic and didn't use a PCB.

I'm wondering if it may be caused by the jumper cables that I use the pins for the SPI connector (as I don't have a crimping tool). As the jumper cables are very thin and generally not really appropriate for soldering etc.

FixyNeko, 2022-01-16:
I had an issue where I mirrored the pins on the connector. On the wiring schematic you're looking at the wheel side, not the base (I definitely feel like this should be highlighted in this project on the schematic, by the way).
Might be a reason for the base protecting itself and shutting down

smollda, 2022-01-18:
As for contacts, these fit pretty well https://www.tme.eu/en/details/7.010.901.001/m23-connectors/hummel/

Berstrpriv, 2022-01-19:
@FixyNeko
Man, reading this gives me headaches again. I thought about mirroring the pinout way too long.
So, you say that I don't have to mirror anything and just connect it as it's written in the schematic? Because I tried mirroring it vertically and after a few reboots I stopped.

FixyNeko, 2022-01-19:
The schematic is looking at the wheel connector.
What you want to do is putting pins in the base connector. So you have to mirror left<->right

Berstrpriv, 2022-01-19:
@FixyNeko
Yeah alright. Good thing I did that. Otherwise I connect 5v to sclk. That would be massive ouch. Still had these reboots. Need to redo the adapter plug.

FixyNeko, 2022-01-19:
I connected it today in my office (was working elsewhere when I got it working). I have massive issues when I connect anything to the base, more precisely the sck pin is very sensitive and the base shut down (red led slowly blinking). This also happens when I connect only myoscilloscope, very strange. The sck, cs and mosi signals are very good however, they are still sent and no weird tension drop is happening

Berstrpriv, 2022-01-19:
Yeah, it would be nice, if some of the guys with the work adapter could send me their adapter to check on my dd. I am willing to pay. But the currently only available emulator is 100€ without the qr. This is just way too much.

bbogdanmircea, 2022-04-25:
Hello, I could not understand from the discussion if this works or not with CSL DD reliably?
Is there a list of all the Fanatec Wheelbases that are supported by this project?

Berstrpriv, 2022-04-26:
@bbogdanmircea
From what I understand, it is possible with the dd, probably also with podium since all wheels work on all bases, the ffb-check-mechanism has to be the same.
Personally, I stopped trying this. It is not worth it in my opinion. Go for the podium hub. It will be worth it.

lshachar, 2022-04-26:
@bbogdanmircea,
All of fanatecs wheels are interchangeable with all their wheelbases. Because of that reason there isn't such list. 
FixyNeko confirmed this works with a CSL DD.

cihaneren, 2022-05-06:
Hello again, I've tried with my CSL DD. I've just needed it for classic steering wheel and I've built the project without buttons or 7 Segment display. Base recognizes Arduino and starts blinking in red. But the driver software of base doesn't recognize the wheel. Should I change something in Arduino code ? Or should I install something more on PC ? (Sorry for noob questions again)

FixyNeko, 2022-05-07:
The base blinks red when an error occured. Did you update the base formware to the latest verzion ? Mine came from factory with a version this project was not compatible with

cihaneren, 2022-05-07:
> The base blinks red when an error occured. Did you update the base formware to the latest verzion ? Mine came from factory with a version this project was not compatible with

Yes, I'm using with my McLaren V2 in latest version. But I've tried to reinstall the firmware manually and it did not help.

lshachar, 2022-05-08:
Connect the arduino to your computer via USB. start a serial monitor. See if the arduino actually gets proper SPI communication with the wheelbase. every line that should be displayed on the LCD screen should also appear on the serial monitor. send C6 on the serial monitor. see if you go into the menu system. read the last comment in [arduino_fanatec_wheel.ino](https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/arduino_fanatec_wheel.ino)

cihaneren, 2022-05-09:
Hello @lshachar I uploaded a screenshot of Serial Monitor but I couldn't understand anything from it. I think there is a problem about this.
[image]

bbogdanmircea, 2022-05-09:
You are not having the same Baudrate set in the Serial Monitor as in the Arduino Sketch.

cihaneren, 2022-05-09:
Now it says "incoming data CRC8 mismatch"... I'll try it with another shifter or arduino :/

Jandako, 2022-08-08:
It does not work for me either on my csl DD...

Balterr, 2022-08-14:
Good evening everyone. Félix FixyNeko How did you run SPI on RP2040 in slave mode?Sorry for my English, I'm from Ukraine.

starter8, 2022-10-07:
Hi. So after bypassing 13pin plug I connected USB C extension to base and connected genuine wheel on the  other side and connected other rim on base. Then I decided to try making emulator and connect it directly via USB port, but I put soldering on hold since pin 3 didn't mach the ground pin. In my findings pin 12 is other ground pin. But decided not to risk it.