lshachar/Arduino_Fanatec_Wheel #18: Understanding the MOSI buffer

jxweric, 2022-02-12:
Hi everyone, I'm currently trying to add a LED strip as a rev display. Therefore, I have to understand which values in the mosiBuf[] contains rpm values. Can anyone help me to understand the MOSI buffer, which cells contain what information, or is there a list of the MOSI bits information so that I can add other support I might need in the future? Any help is greatly appreciated!!!

Have a great weekend :)

lshachar, 2022-02-13:
[Here's what you're after](https://github.com/darknao/btClubSportWheel/blob/master/dev-tools/cap.py)

As said on the readme file, this project would never have been possible without the work of Darknao on his [BtClubSportWheel - converting any Fanatec CSW steering wheel to a standalone USB version](https://github.com/darknao/btClubSportWheel) project.

StuyoP, 2023-05-29:
Did you made any progress on this? I am doing the same, already added support for rotary multiposition switch and matrix keypad to save some inputs. Already added LEDs but only for backlight, am looking currently to incorporate REV coming from base. It is quite complicated for me to understand the btClubSportWheel project, I see all the information in there but cannot compile it. Also am trying to add support for an APM module - I have a 3D printed design that works very good with a Podium Hub, but I want it supported in emulator to unlock the full potential for my new wheels :)