lshachar/Arduino_Fanatec_Wheel #19: Teensy 3.2

meggerja, 2022-02-13:
Hello, sorry for maybe stupid question but I would like to ask is this OK to use teensy 3.2 as it's operates at 3.2v instead of arduino with logic shifter?

lshachar, 2022-02-13:
I can't give you a straight answer. I tried to read back on [this huge issue that has lots of good info](https://github.com/darknao/btClubSportWheel/issues/12), and I searched for info in my github, but didn't get anything concrete. If the teensy works properly as an SPI slave, and can keep up with the base clock of the SPI bus at 500KHz, then in theory it should work. But not all boards are born equal and I never tried working with a teensy. Didn't see anyone mentioning getting my code to work with a teensy thus far.

[Sabifa got it to work with a Pro Mini V2](https://github.com/lshachar/Arduino_Fanatec_Wheel/issues/9).
That board can be run at 16MHz at 3.3V and it does work at least to enable FF effects, but I found out that the Pro Mini is not really designed to run at 16MHZ / 3.3V, so I decided not to fully support it here.

Good luck

meggerja, 2022-02-14:
Hello
I found this:
https://www.thingiverse.com/thing:3673632
Do you think it may work?

lshachar, 2022-02-14:
Yes

meggerja, 2022-02-14:
I'm not sure what do you mean with this center point calibration, this is available threw the pc fanatec software, or maybe I'm thinking about something different 🤔 I've got csl dd so might be different

Kuillemaul, 2023-10-07:
Has anyone made code for the teensy 4.0 as the 3.2 is no longer available.?