lshachar/Arduino_Fanatec_Wheel #20: Wrong schematic?

Shane87F3Ng, 2022-03-09:
Hi Ishachar,

is it possible, that the schematic shown at 

https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

is not correct (anymore?)?

Because why you connect the 3V3 from the Base to the 3V3 connector on the arduino, but in your detailed schematics, this connection line is not shown. So which one is right? :)

Cheers Shane

lshachar, 2022-03-13:
Hi Shane,
There is no need to connect the 3v3 on the arduino nano to the low level side of the logic level shifter. Thanks for noticing that. (It was a different user who've made that schematic in the first place!) I will upload that file.

[This is the correct schematic](https://raw.githubusercontent.com/lshachar/Arduino_Fanatec_Wheel/master/Schematics%20%26%20pcb%20layout/how%20it%20looks%20like.png)

Shane87F3Ng, 2022-03-13:
Thanks for your quick response Ishachar :)

A further question, have you ever tried to power your nano board with the 3v3 line from the base to the 5v pin of the arduino? Because at that pin, there is no voltage regulator followed, so digital pin output is also at 3v3. Most of the boards (CH430 clone versions) are actually capable of running at 16mHz at 3v3, as i am using it like that on my Thrustmaster wheel. The positive effect on that is, that no logic level shifter is needed and getting rid of all the noises cause by it.

But i am not sure if the 5v line of the base needs to be used in order to get the wheel recognized

Greets
Shane

lshachar, 2022-03-13:
I've never tried running it like that, (and I suppose that the processor isn't ment to run at 16mhz at 3.3v), but anyways - once you connected the nano to usb, the outputs will go up from 3.3v to 5v. then you're feeding 5v to 3.3v inputs on the fanatec wheelbase side.

Shane87F3Ng, 2022-03-30:
OK, yeah tomorrow i will have time to solder everything together. Does anybody know, If i need all 13 Pins or can i just go with 8, that are actually used?

lshachar, 2022-03-30:
...What would you solder all the unused pins to?
Obviously you don't need to solder them.

Shane87F3Ng, 2022-03-30:
I mean not to solder them, but are they necessary in a way, that the base can detect the pin contact. otherwise i would just go with 8 Pins, you know what i mean?

closed 2022-04-06 by Shane87F3Ng

ProjectCoops, 2022-04-19:
Hi Ishachar great project, i have and original pcb csl elite p1, i would like to cut the pcb board down for a drift wheel and require no buttons,
1. MISO
2 .MOSI
3. GND
4. 5v
5. GND
6. SCK
7. 3v3
8. CS
Could you please tell me which of the first 8 pins are necessary for the wheel to be detected for ffb, i can calibrate via fanatec properties.
Thankyou.

lshachar, 2022-04-19:
All of them.