lshachar/Arduino_Fanatec_Wheel #21: >Implement Buttons and encoders

Shane87F3Ng, 2022-03-09:
Any help / hints on that?

Hello @lshachar,

first of all, thx very much for your work, that is just awesome!
I have one question about how to add additional buttons / encoders..
Do i need to change to another wheel than the Porsche RSR? or just modify the button matrix and bytes?
Because as far as i know, i can only use buttons that are originally supported by the wheel? it is like that with thrustmaster

Actually i build sth similar for my previous thrustmaster base and was reading the wheelstates with an arduino sketch by https://rr-m.org/blog/ to implement it into my sketch and simulate the right wheel..

my plan is to use:

3 encoders  + push button
8 buttons

so in total i would need to have 19 buttons (wheel states)

any help on where to start?

@EtienneGameSeed As is saw you do not sell the teensy anymore, maybe you can help with some thoughts here without getting too much info away on your own project.

_Originally posted by @Shane87F3Ng in https://github.com/lshachar/Arduino_Fanatec_Wheel/issues/12#issuecomment-1060968563_

Shane87F3Ng, 2022-03-12:
OK, i figured it out :)

lshachar, 2022-03-13:
I think your assumption will turn out to be true - you should figure out which of the fanatec wheels supports all of your wanted 19 buttons (check out the fanatec universal hub as well), because it sounds feasible that any wheel will only support buttons that should physically be on it.
once you have your answer, it's very easy to change the wheel byte in the arduino code, as you probably have found out by now.
Start off with changing some button bytes and noting that the button states in fanatec's driver (or in a game / other joystic utility) is actually changing.
The rest should follow easily!

Shane87F3Ng, 2022-03-13:
Nice. I will try a setup with a button matrix and report back as soon as i have results :)

lshachar, 2022-03-13:
At first you don't even need a button matrix. There's a way to toggle bits on and off using the serial interface. look it up in the code and try flipping some bits that are used for buttons. they should toggle in the controller driver application