lshachar/Arduino_Fanatec_Wheel #22: Holding pressed any button it operates repeatedly

fsgroppo, 2022-07-22:
Dear Ishachar

First of all my congratulations for this project, your work makes this community benefit by adding value and facilitating access to knowledge for all of us who started in this type of project. Congratulations again.

Issues = I followed your instructions to the letter and put together this adapter for a Fanatec CSW 2.5 base. It works almost flawlessly except for two details;

a. when the adapter is in use (after base calibration, up and running) the TM1637 display correctly shows the chosen gear (in game) in the second segment, but randomly and for a very brief moment it also shows in the third segment; there seems to be some problem sending data from the arduino to the TM1637 display. This happens all the time, regardless of the information that the display must show; in fact it doesn't matter if you are in a game or not, the display shows a blinking in the third segment randomly.

b. I have found that by holding pressed any of the gear shift levers (I use the arduino A1 and A2 outputs) the gear changes occur consecutively in ascending or descending order depending on which lever you press; example: I press and hold the lever to shift up and the gear shifts from neutral to 1 and after a very short moment it continues to shift up 2, 3, 4, 5, and 6 (without releasing lever) when it should just shift from neutral to 1 no matter how long you keep the lever pressed. This operation that I describe causes that when you are racing inadvertently two consecutive gears are raised or lowered.

So my question to you is if it is possible to change any parameter or line of code to resolve a. and b. (especially b., which is the most important). I have no programming knowledge, but if you could tell me which line of code and which values ​​to test I would be closer to finding a solution to these issues.

Thank you, greetings from Argentina.

lshachar, 2022-07-22:
Hi @fsgroppo, and thanks for the kind words :)
connect your arduino to your computer via USB and start a serial monitor. I'm guessing that you will see "incoming data crc8 mismatch!" warning messages popping up on the monitor. This explains why your gears are bouncing as well as the display faulting.
If I'm right, please look around in the other opened and closed issues here about how to fix the crc8 mismatch problem. basically, it's difficult to get a good SPI connection between the wheelbase and your arduino because the wheelbase is very sensitive to interferences.

fsgroppo, 2022-07-22:
Thanks for your answer. I attach what I get through the serial monitor (it's the very first time I use it). I don't understand anything at all, any suggestions?

Greetings
[image]
.

lshachar, 2022-07-25:
Change the serial baudrate to 250000.

fsgroppo, 2022-07-25:
Dear Ishachar.

Thank you for your reply. Now I have managed to get more consistent information, I attach the image in case you see something that is relevant and I can not interpret. You were rigth about CRC8 errors; occur persistently without the need to press any button. I have read about each of the problems exposed here and the way to solve my problem would be to try to correct the possible interference problems that you have mentioned. Doing this implies manufacturing a completely new adapter in my case because I have strained with epoxy resin all the components (conductors, arduino, logical level, etc.).

Then... I go with my last attempt to consult you whether or not it is possible to do something or try something different with the code, of course I repeat I do not know what statements like these do:

delayMillis = 400
dataLength 33
lastPrintMillis = 0

but if it occurred to you to try something different with these values or explain to me what these values define I could make one last attempt before discarding the adapter I already made and having to make a new one directly addressing the necessary changes to avoid interference.

Thank you very much for your attention and take care to dispel my doubts.

Greetings
[image]
[image]
.

lshachar, 2022-07-26:
Your adapter looks really really nice!
The code isn't the issue, and there should be nothing that you could change that would help..
It's a game of having large ground pads everywhere possible, and good contacts on all your SPI signals, But it's just not easy to get it working very well.
You can try using a genuine arduino as well, that might help.

One option that comes to mind is - use another arduino (such as arduino pro micro) to connect all the buttons to. run its line outside of the steering wheel with a flexible cable. have it connected to your computer.
Having that second USB cable is kind of annoying, but also trying to get good SPI communication with the wheelbase...

fsgroppo, 2022-11-09:
Dear ishachar

I have made a completely new emulator (all new hardware too) following your advice, I am attaching some photos illustrating the work. Unfortunately I still have the exact same problem. There is no other thing else that I can do. I plan to try to remove from the code - and I don't know exactly how to do it, it will be trial and error - the features that I don't need just to make the code shorter, but I'm sure it won't solve the problem either. when I say remove from the code I mean not to comment the line of code with //, but to delete it from the code completely. I don't need the display, I don't need the dpad. all I need is x2 buttons to actuate the gear paddles. After this... it will be time to leave the project with a bittersweet taste if it does not work. I congratulate you for the work you have done. A big hug from Argentina.

[image]