lshachar/Arduino_Fanatec_Wheel #25: Other wheels reference IDs

Alexbox364, 2023-04-15:
Hi everyone, 

first, thank you very much for sharing this very good code basis to interact with Fanatec bases!  I had it up and running quite fast and SPI implementation is great!

I'm working on a project to build a custom wheel based on a Podium hub and the fantastic Turn 911 GT3 R 2023 DIY wheel (freely available build up files) (custom wheels based on USB cables or Bluetooth communication did sound a good option to me)
https://www.thingiverse.com/thing:5702348.

As a first step, I tried to find out which Fanatec wheel to replicate to have so many buttons so I tried to identify other wheels and their associated ID in the SPI protocol. Here is the extended list I build :

{0x01} for BMW M3 GT2, 
{0x02} for ClubSport FORMULA, 
{0x04) for Uni hub
{0x05} X
{0x06} universal hub hub xbox one
{0x07} CSL elite p1 xbox
{0x08} CSL P1 v2 - FW update required
{0x09} Maclaren GT3 v1
{0x0A} clubsport formula v2 / v2.5
{0x0B} Maclaren GT3 v2
{0x0C} Podium hub
{0x0D} not supported
{0x0E} Bentley - FW update required
{0x0F} Podium M4 GT3
{0x10} X
{0x11} CSL universal hub
{0x12} CSL elite WRC 
{0x13} BMW GT3 v2
{0x14} X
{0x15} CS universal hub v2
{0x16} CS wheel f1 esports v2
{0x17} Podium BWM m4 GT3
{0x18} and above - nothing

I discovered by the way the so good looking Bentley Fanatec wheel!

For some wheels, my DD1 base required a wheel firmware upgrade to work.

I hope it can help others build their own wheel. I'll post updates here if I can help others.

cheers

StuyoP, 2023-05-22:
Hi,
Thanks for this list, it will help me greatly in my next project!
I actually already build the Bentley wheel, but am driving it with USB and SimHub :)
Did you make any progress on the other wheels? I am looking into building something that support rotary positional switch (something like the F1 toggle rotaries) in order to use as MAP dial. Also wondering if APM module is possible to be interfaced as I already build custom 3D printed one that works great with the original Podium Hub.

Thanks!

StuyoP, 2023-05-30:
Hi mate,
Thank you very much as well! The servo one is great, but I had a servo die on me after a year of heavy use, switched to the SimHub native stepper motor one (as myself also cannot anymore without belt tensioner). Working on video and written review ATM, feels better and is quieter.

About the Fanatec emulator - I moved quite ahead already. I made it work powered from base, added keypad matrix to save some pins, added rotary switch voltage dividers and translated them in code to button press, added initial aRGB support. I am currently wrapping up the Ford Puma WRC Wheel button plate and all three rotaries work, together with 9 buttons. Have total of 13 buttons coming from default profile and am 1 button short so wanted to move to another profile. Substituted the OLED screen with a 3 digit 7 segment, looks pretty good for the time being.

Already ordered some ATMEGA329P chips and will make a barebone version as well at some point and will hopefully make it work at 3.3V to ditch the Level Converter. I suck badly at programming and struggling currently to implement dual clutch so I can add my APM module and make the Bentley wheel natively supported. I can incorporate the LED matrix to show speed but for the most advanced features it would not be possible to make them native. 

I went through the original emulator code - most of it is already there. Also clutch, additional paddles, REV LEDs, and one rotary - this is the standard supported SPI, but as I said - I suck at programming and have issues isolating the code and transferring it here.

I am documenting everything and will make extended material how all was assembled. Pretty happy with this project, it unlocks quite a potential for all my other wheel projects.

As far as the other "advanced" features go - they are not part of the standard SPI and I doubt they are there. Most probably the are additional controllers which communicate through the additional pins (the ones for the advanced Porsche module) or additionally encoded somehow in the original SPI package (less probable option). I have the basic skills to start monitoring and decoding, but not much free time as am pressing the other projects. In general if this is supported by the reverse Fanatec emulator which is the base for the code here it should be pretty straight forward to extend.

You can find me in the Discord channel of the site where we can discuss further. I have another member there and we are struggling together to push forward this project :)

Hope together we will manage to extend it even further!