lshachar/Arduino_Fanatec_Wheel #26: CSL DD wheel base connects and discconects with wheel attatched.

thomaswang04, 2023-08-15:
Hello everyone, sorry if this is a stupid question but i was able to get the wheel wired up the only way i've  been able to get it to work is by having the Arduino powered up before turning the wheel on. Otherwise when the wheel starts up it instantly disconnects after start.

lshachar, 2023-08-15:
It's not a stupid question, but it is lacking information. I'm not sure if your arduino is properly powered from the wheel once the wheel starts. If you followed my diagram, there's a diode so that you're not powering the arduino from USB while also powering it from the 5V pin connected to the wheel. the diode is there so that the power supplies on the wheelbase do not get voltage from the PC's power supply.

mbakos82, 2023-08-28:
Hi. I have the same problem. When my Arduino is plugged in to USB before I plug it into the wheelbase, the steering wheel is detected. Otherwise the wheelbase disconnects. I also found out that when there is no wheel plugged into the wheelbase, I measure voltage across the 5V pin and any of the GND pins, there is no voltage at all. Maybe the wheelbase powers on the 5V pin after SPI comm is established. It might be something else totally. My test scenario is a Nano with level shifters and a bush of wires on a breadboard.

lshachar, 2023-08-28:
That's an interesting find! see if the 3.3v line is always supplied once you turn the wheelbase on, and see if the 5v line is off at the same time. then see if the 5v line turns on after spi comms is established. 
(have the arduino running from an external 5v supply, and have a common ground connected (between the arduino and the wheelbase. then you can measure the 5v line externally with a multimeter to see at what time it turns on)

If the wheelbase behaves the way I described, then it means that *all* of fanatec's wheels must be running from the 3.3v line, with the 5v probably only there for peripherals (or never used). and it would also mean that anyone with a CSL DD wheelbase would either need to add a 3.3v to 5v step up converter (boost converter) to power the arduino nano from, or completely change the nano for a 3.3v board. There's some information about this in other issues and [in this pull request](https://github.com/lshachar/Arduino_Fanatec_Wheel/pull/13)

mbakos82, 2023-08-28:
There are some more crazy things happening in the wheelbase without a wheel connected.
After turning on the wheelbase (no wheel connected), I measure 3.16V on pin 3 compared to pin 5. (Both should be GND.) On pin 7 (3.3V pin) I only measure 0.3V. Then I connect pin 3 and pin 5 through a 10 ohm resistor, 2 things happen. 3.3V comes alive on pin 7 and the wheelbase disconnects. Pin 7 remains live after I remove the resistor. Pin 4 (5V) comes alive as well.

I tried connecting the wheel with arduino 5V power only. The wheelbase 5V pin comes alive immediately when I connect the arduino to SPI.

thomaswang04, 2023-08-28:
Could it be that the fanatec wheels are getting power from one of the other pins that are unused? I looked online and it looks like pins 9 and 12 are used as a usb pass through for power and ground. I will try and check those pins when I get the chance

mbakos82, 2023-10-06:
Hi again. I have come across a new theory after letting my frustration settle a bit. On the original wheel connectors, one pin is shorter than all the rest. Has anyone wondered why that could be? My theory is that when you connect the wheel to the base, the GND pin on the outer rim is connected a tiny bit later than all the rest of the pins. It is when the 2 GND pins are shorted inside the wheelbase, SPI packets are beginning to be sent to the wheel. By fully plugging in the adapter, that very short timeframe could be required to boot up the adapter to a state that it is ready for SPI communication.
I have taken my adapter apart already, I haven't had the chance to test this idea.

I would be interested in how many people have taken care of making that one pin shorter and how does it relate to the success of their project.

lshachar, 2023-11-09:
I remember starting to reply but apparently never sent it. No I would not think that the shorter pin is relevant. these are pogo pins (spring inside) and maybe it got stuck pushed in. or a manufacturing defect. a drop of wd-40 might release it. however I'm rather sure it's not intentional or serving a purpose.

@thomaswang04  everything is possible, I researched a different wheelbase when the CSL DD did not exist yet.
However, since my arduino wheel works very reliably on my CSW V2.5, I'd expect it to work the same on other wheelbases too.

irisnotaprogrammer, 2023-11-09:
> @mbakos82 I remember starting to reply but apparently never sent it. No I would not think that the shorter pin is relevant. these are pogo pins (spring inside) and maybe it got stuck pushed in. or a manufacturing defect. a drop of wd-40 might release it. however I'm rather sure it's not intentional or serving a purpose.
> 
> @mbakos82
> 
> > There are some more crazy things happening in the wheelbase without a wheel connected.
> 
> I did point you in a direction of testing the 3.3v line, see my previous comment, you never said if you tried that.
> 
> @thomaswang04 (...) However, since my arduino wheel works very reliably on my CSW V2.5, I'd expect it to work the same on other wheelbases too.

I can confirm this works on a CSL DD GT Pro as well.  Succesfully made this project for mounting a truck wheel to my Fanatec base.

For some information: I used an Arduino Uno with a level shifter, because that was what I had laying around, except for the level shifter :P .

Base powers up as it would with an actual Fanatec wheel.

As for there being pins that might behave different: given that there are wheels with OLED screens in them, I suspect that they might act like a small DisplayLink device and need USB communication that way.

irisnotaprogrammer, 2023-11-21:
> ....and just stop having all sorts of weird wiring issues...
If you would see my home-made connector on the Fanatec base for extending it to the truck wheel along with what the test setup looks like you would understand that it's probably down to people misreading the schematic or not providing good joints (I am on a Discord server where people try to use taped together wire joints...)

> btw I'm quite disappointed with how nobody respects the fact this project is a donationware. :/

Let me make that up ;)

By the way, are button matrixes possible?  I am currently not exactly out of button pins and the addition of buttons that there could be would be up to 4 extra of the 5 I already use so there are definitely pins left on the Uno, but I was simply curious.

thomaswang04, 2023-11-21:
> By the way, are button matrixes possible? I am currently not exactly out of button pins and the addition of buttons that there could be would be up to 4 extra of the 5 I already use so there are definitely pins left on the Uno, but I was simply curious.

I assume as long as you can program it it will probably work

lshachar, 2023-11-23:
@irisnotaprogrammer 
> Let me make that up ;)

This made me very happy, thank you! I really appreciate it :) 

> By the way, are button matrixes possible? I am currently not exactly out of button pins and the addition of buttons that there could be would be up to 4 extra of the 5 I already use so there are definitely pins left on the Uno, but I was simply curious.

Yeah as @thomaswang04 said, it's just a matter of programming it, and it should be very straightforward. the only limitation would be the size of the button byte array, which is either 22 or 24 buttons. (as the last comment in [arduino_fanatec_wheel.ino](https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/arduino_fanatec_wheel.ino) says)

> (I get that it might feel uncomfortable, but you could hide a part of your project behind a payment. It's definitely not foolproof, but still).

Nah, not for me. thanks for the thought though! :)