lshachar/Arduino_Fanatec_Wheel #27: CRC8 Mismatch and low voltage on L1/H1 pins

minnis23, 2023-09-27:
Hi, I'm attempting this project without the screen or buttons (I've got another solution for this).
The wheel isn't working, and opening up the serial monitor I am getting the CRC8 mismatch error.
I have looked at my connections with a multimeter - on both sides of the logic level shifter, I'm getting 3.3v and 5v for all pins except L1 and H1, where I'm getting about 0.2v. I presume this is the issue, or is this pin meant to be low?

lshachar, 2023-09-27:
You are reading digital lines with a multimeter. Which tells you nothing. If you want to see whether the data is really transmitted properly - you need an oscilloscope.

minnis23, 2023-09-27:
Okay - I thought that the point of the logic level shifter was to transform 3.3v inputs to 5v and vice versa, as it is with the other pins - is this not the case with L1 and H1 then? I don't have an oscilloscope - is there any other way to confirm using just a multimeter, even if it's ruling out that all _other_ connections are okay? All LV pins show 3.3v and all HV pins show 5v (bar LV1/HV1 as mentioned above). I had initially forgotten to add a 5v line to the arduino but I have corrected that and the arduino is now getting 5v. Of the 4 pins on the arduino, the power and TX LEDs are bright, the RX LED is off, and the "L" LED is showing vague signs of wanting to flash but isn't properly lighting up. Does this point to anything in particular? Apologies if any of these are rather silly questions - I've done several projects with arduinos before, programming them through Simhub, but this is the first time I'm attempting something more complex.

minnis23, 2023-09-27:
Just for extra info. 
At no point does Fanatec control panel recognise the wheel.
If the wheelbase is turned on, I will hear the Windows "disconnect sound" as soon as I attach the wheel and the wheelbase will disconnect. Simply removing the wheel isn't enough to sort this, I need to remove the wheel and turn the wheelbase off and then back on.
If the wheelbase is turned off, upon turning it on I'll be met with the Windows "connect sound" followed immediately by the "disconnect sound" and the wheelbase (but not the wheel) will momentarily flash up as connected in the Fanatec control panel before disappearing again.

bananafrog, 2023-10-07:
Hi, did you end up finding a solution? I have the same problem on my pro micro. I'm going to try wrapping a ground wire around the data pins, as that what I've seen on other threads