lshachar/Arduino_Fanatec_Wheel #28: Wheel keeps disconnecting in Control Panel and no Buttons etc. work

janmannsbart23, 2023-11-06:
Hey all,

I'm currently working in an STM32 implementation.
I'm using a SPI DMA Slave for communication and wanted to try in the first place if the wheel is detected in the software and button presses are recognized.

I've used the data paket as mentioned in your sketch.

The wheel is also recognized in the Fanatec Software, but periodically disconnects for a second and then reappears.
I also tried to set some button bits (like in the image), but nether of those were recognized in the control panel.

I've captured the communication with an analyzer, but as far as I can see and understood the protocol, this should be fine

[image]

I also cannot see any abnormalities watching the communication over a longer time, that might explain the short disconnects of the wheel. 

Any help would be very much appreciated.

janmannsbart23, 2023-11-06:
For completion:
Im on 
CSL DD
PC Driver 451
Wheel Base: 1.1.85
Wheel Base Motor: 1.0.3.2

lshachar, 2023-11-09:
Hi @janmannsbart23,
See if you sometimes get a text on the alphanumeric display data bytes. I wonder if the text is something like 'UPD' which means that the wheelbase is trying to get the wheelbase to update, which it cannot with my code.
For me (CSW V2.5) when that text appears it flashes, so it won't appear in all data bytes, so set up your way for catching it appropriately.
For me it's not an issue and the wheelbase doesn't try to update the wheel every time, only once every idk 20 times I plug in my controller. but I'm wondering if the CSL DD tries to automatically disconnect and reconnect the wheel to update. and then you might be getting the disconnects. just a thought.

There is something fishy with the CSL DD. if you go through the issues tracker you'll find quite a few issues relating to that wheel. you owning a scope and actually knowing your way around SPI coms, puts you in a great place to actually solve the problem. I think you might gain some insights from other people's posts about the CSL DD.