lshachar/Arduino_Fanatec_Wheel #30: Confirmed working: CSL GT Pro, Arduino Uno and level shifter

irisnotaprogrammer, 2023-12-04:
Ischachar, in case you don't want confirmed builds for wheelbases other than the one your guide focuses on you can close this or remove it since it's not an issue.

For anyone with a CSL GT Pro and is wondering whether or not this tutorial works for that wheelbase: it does.
I have made one with an Arduino Uno, and a level shifter.  The level shifter piggybacks on the Arduino's header pins and uses a DIY'd piece of lane PCB for this, as you can see in the pictures.

It carries 6 direct buttons, and a D-pad with a center-contact (pushing down) that came from an old television, a D-pad I hope to use to navigate game menu's with.
This is all meant to make the wheelbase have Force Feedback because of the Volvo truck wheel attached to the base (through a piece of steel pipe going through a barrel-ball-bearing assembly 50 cm's away from the wheelbase quick connector) because the truck wheel weights around 2,5 kg which I considered to be a bit much).

5 Of the buttons on the Arduino are the 4 cruise control buttons (cruise control on, resume on the left, and increment and decrement the speed set on the right) and the horn button (interconnected, I have thought about making them distinct in order to trigger different types of horns for the games that have this support)
The D-pad is on the underside of the right cruise control button module, and was not originally there in the wheel.

[image]

The D-pad required changing the line in code from INPUT to INPUT_PULLUP because values where otherwise to close together.
[image]

wegreenall, 2023-12-13:
I have been looking at doing this for my CSL DD as well. However there seem to be lots of pinout diagrams and it's a bit confusing. Given that you have got this to work, could you clarify the pinout of the wheelbase so I can work out exactly what to plug in where? my 5v seemed to be on a different pin and of course I might just be being dumb about it, but it'd really help to get clarification from someone who has definitely done it on a CSL DD.

irisnotaprogrammer, 2023-12-16:
Can I ask you if you happened to make the mistake of not mirroring the pinout?  Because I probed it with a scope and it is exactly the same as it is in the tutorial that Ishachar made, but a common mistake to make when not working electronics much can be to forget to mirror any images.
(And even if you are experienced, missing the "PIN LAYOUT - BOTTOM VIEW" in a datasheet can cause you to scratch your head for hours until you realize you were not looking at some of the pins in the required order).

When I looked at this picture and the left connector:
https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

I found out the connector inside my Fanatec wheel (the simplest you get with one of the bases, the previous owner probably used a different base before switching to this one) has 1 pin extra and while unpopulated, the pinout seemed to differ a bit (difficult to know for sure since it has all kinds of interconnections with capacitors, so I didn't trust that to much during probing with a scope).

As a result I followed the pinout on the right part of the picture, and made sure to mirror it (since you will be working on the back of it basically, so it is practical to have the pinout like you are looking at it, and not from the side you can't see when plugged in.
I followed the drawn diagram (I will add  it can throw you off easily due to the mixed colors and mixed numbers especially on the left side of the picture holding the surface-mount angled connector) behind the level shifter to connect the right wires to the right inputs on the Arduino, and got it to work that way.

So best to check first: did you happen to measure 5V on what appeared to be pin 5 because you didn't flip the picture?

(And before you want to feel stupid: my brother designed and etched a whole PCB once for a power supply he was making, and etched the control chip with all the pins inverted for thinking he was having a top-down view during design, and because he get's confused when doing bottom-layer etching toner-transfer prints, he tends to accidentally flip his design).

wegreenall, 2023-12-16:
I thought i had but perhaps I was just getting confused - i just thought i
would ask for confirmation! I will use that diagram in that case. Thanks!

irisnotaprogrammer, 2024-01-09:
The code by the way allows combining libraries!  In my project, the Volvo wheel that I use has horn buttons in thumb-positions at 9 and 3 o clock.  My hands are a bit on the small side to reach them easily, and I have found that trying to rotate my hand inwards toward a horn button while trying to also keep in contact with the wheel makes me go in a different direction more easily than just taking my hand off and pressing the middle of the wheel (which doesn't have a button).

I decided to use the ADCTouchSensor library by Alexander Pruss to turn the metal Volvo logo into a horn button:
[image]

I unfortunately had my hand over the microphone of my phone, so it is difficult to hear, but here is the "button" in operation:
https://github.com/lshachar/Arduino_Fanatec_Wheel/assets/143168797/33b623da-c032-4317-bc88-c264d340d14b