lshachar/Arduino_Fanatec_Wheel #32: Yet another CRC8 issue

johnysed, 2024-04-27:
Hi, so Im doing just the ffb unlocker and I can't get it to work. 
There Is like 98% CRC8 mismatch lines. I shielded everything, checked the connection, etc. Any other possibility?

Im using arduino Pro Mini without bootloader. Connected directly, as it is 3,3V. 
Everything is connected as in your schematics, except for the 5V pin from the base, I don't connect that at all.

I managed to get it running like twice, but no luck otherwise.

Thanks for any help

lshachar, 2024-05-01:
way to go for getting it working!
what was it that resolved your crc8 mismatch issue?

I can't tell you what's not working. did you get to try it with a proper fanatec steering wheel?
Open the run command window:
win+r
type:
joy.cpl
do you see your fanatec joystick there?
double click it
do you see a steering wheel connected or just a wheelbase?
can you start the ffb test or is it not available?

oh, now that you have crc8 matches, try to comment all serial communication messages and see if that makes a difference (in case you still cannot use your wheelbase yet)
I've had an issue with slow response from buttons (that are installed on my rim) and that was the culprit.

johnysed, 2024-05-01:
Thanks for your comments, it seems like the 3V3 8mhz is not enough, or not sure. 
I did your version with 5V 16MHz and logic shifter and that works flawlessly.
I'll try to poke in it again, but probably will just go the route of the logic shifter.

lshachar, 2024-05-05:
oh yeah, I wouldn't expect it to work at 8mhz. github is hiding your edits very well, I didn't see that information before. you can try going the stm32 way, to get 3.3v & over 16mhz in one board.