lshachar/Arduino_Fanatec_Wheel #34: Hi can someone tell me which drawing is the right drawing to make the nano emulator with plz

J3161, 2025-01-04:


lshachar, 2025-01-05:
This is the full schematic 
https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/Schematics%20%26%20pcb%20layout/Legacy%20layouts/how%20it%20looks%20like.png

and this version just to get ffb working
https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

J3161, 2025-01-05:
Hi ishacher  so this will work on ps5 when I make the emulator I just want access to all buttons to create my own wheel

J3161, 2025-01-05:
Hi ishacher if you can explain me In Step by step on how to make the nano emulator I will pay you petrol money

J3161, 2025-01-06:
> This is the full schematic https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/Schematics%20%26%20pcb%20layout/how%20it%20looks%20like.png
> 
> it's stored here https://github.com/lshachar/Arduino_Fanatec_Wheel/tree/master/Schematics%20%26%20pcb%20layout in the KiCad folder
> 
> and this version just to get ffb working https://github.com/lshachar/Arduino_Fanatec_Wheel/blob/master/schamtics%20arduino%20nano-level%20shifter-fanatec%20round%20plug.png

Hi ishacher I got it working with the fanatec ffb diagram but emulator keeps glitching with numerous buttons. I want to use buttons is there another code I need to download or is your Arduino ino file the right code. Thanks